emericg/SmartServoFramework #2: Compilation error: undifined ref to `dev_testlock', `dev_unlock', `dev_lock'

romainreignier, 2015-08-19:
Hi!
First of all, congratulation for your nice job on that framework!

I did use the Gui some months ago and it worked well. But today, I have to use the latest changes and the compilation failed.

I am using Ubuntu 14.04 with GCC 4.9.

While building the GUI, I have the following errors:

    [...]
	build/SerialPortLinux.o: In function `SerialPortLinux::isLocked()':
	SerialPortLinux.cpp:(.text+0x4f9): undefined reference to `dev_testlock'
	build/SerialPortLinux.o: In function `SerialPortLinux::removeLock()':
	SerialPortLinux.cpp:(.text+0x5e0): undefined reference to `dev_unlock'
	build/SerialPortLinux.o: In function `SerialPortLinux::setLock()':
	SerialPortLinux.cpp:(.text+0x949): undefined reference to `dev_lock'
	build/SerialPortLinux.o: In function `serialPortsScanner(std::vector<std::string, std::allocator<std::string> >&)':
	SerialPortLinux.cpp:(.text+0x194f): undefined reference to `dev_testlock'
	collect2: error: ld returned 1 exit status
	make: *** [build/SmartServoGui] Error 1

If I build only the library using CMake, the building is ok but then, if I want to build an example, I have the same errors:

	$ g++ ex_basic_test.cpp -L../build -lSmartServoFramework -std=c++11 -lpthread
	../build/libSmartServoFramework.so: undefined reference to `dev_testlock'
	../build/libSmartServoFramework.so: undefined reference to `dev_unlock'
	../build/libSmartServoFramework.so: undefined reference to `dev_lock'
	collect2: error: ld returned 1 exit status

Maybe my `g++` command is not suitable. I am not a C++ guru... A CMakeLists.txt for the examples would be nice :)

Then, if I try to build with `scons`, I have other errors:

	src/SerialPort.o: In function `SerialPort::setLatency(int)':
	SerialPort.cpp:(.text+0xa8): undefined reference to `MiniTraces_print'
	src/SerialPort.o: In function `SerialPort::checkBaudRate(int)':
	SerialPort.cpp:(.text+0x36b): undefined reference to `MiniTraces_print'
	SerialPort.cpp:(.text+0x3f9): undefined reference to `MiniTraces_print'
	SerialPort.cpp:(.text+0x465): undefined reference to `MiniTraces_print'
	SerialPort.cpp:(.text+0x49d): undefined reference to `MiniTraces_print'
	src/SerialPort.o:SerialPort.cpp:(.text+0x4e4): more undefined references to `MiniTraces_print' follow
	collect2: error: ld returned 1 exit status
	scons: *** [build/ex_advance_scanner] Error 1
	scons: building terminated because of errors.

I am sorry, I have no much time now to try to fix that so I just wanted to make you know of that.

Romain

emericg, 2015-08-20:
The scons / minitraces stuff is an oversights from me, I'll fix that.

You need to add '-llockdev' to your g++ command. The locking stuff can be set up to use lockdev library, basic "file locking" (creating a lock in /tmp/ttyUSB0) or even fopen lock with flock (but this doesn't really work at the moment).
I think I'll default the basic file locking (with no dependencies), because the need for lockdev library to build a simple exemple is not very optimal.
But lockdev can provide some very nice features, like "cross application" locking of the serial port (and not only two instances of the framework).

romainreignier, 2015-08-20:
Ok, thanks for the reply.
I should have though to add the `-llockdev` flag as I had to install the lib...
As I said, I am not use to using g++ directly but Makefiles. My bad.

But for the qmake command, you will have to add it.

emericg, 2015-08-20:
The missing minitrace file has been added. The GUI qmake file already has the lockdev dependencie. I will improve the documentation about that.

romainreignier, 2015-08-21:
Ok, thank you for your reactivity!

But there something I don't understand. The `scons` only build the examples and the `cmake` only build the library. Is that normal?

Anyway, with `qmake-qt4`, the `-llockdev` is indeed added but the building failed with that error. I do not understand why. (`gcc-4.9.2`)

	[...]
	g++ -m64 -llockdev -Wl,-O1 -o build/SmartServoGui build/ControllerAPI.o build/ControlTables.o build/Dynamixel.o build/DynamixelController.o build/DynamixelSimpleAPI.o build/DynamixelTools.o build/HerkuleX.o build/HerkuleXController.o build/HerkuleXSimpleAPI.o build/HerkuleXTools.o build/minitraces.o build/SerialPort.o build/SerialPortLinux.o build/SerialPortMacOS.o build/SerialPortWindows.o build/Servo.o build/ServoAX.o build/ServoDRS.o build/ServoDynamixel.o build/ServoEX.o build/ServoHerkuleX.o build/ServoMX.o build/ServoXL.o build/Utils.o build/main.o build/mainwindow.o build/advancescanner.o build/qabout.o build/settings.o build/moc_mainwindow.o build/moc_advancescanner.o build/moc_qabout.o build/moc_settings.o build/qrc_resources.o    -L/usr/lib/x86_64-linux-gnu -lQtGui -lQtCore -lpthread 
	build/SerialPortLinux.o: In function `SerialPortLinux::isLocked()':
	SerialPortLinux.cpp:(.text+0x4f9): undefined reference to `dev_testlock'
	build/SerialPortLinux.o: In function `SerialPortLinux::removeLock()':
	SerialPortLinux.cpp:(.text+0x5e0): undefined reference to `dev_unlock'
	build/SerialPortLinux.o: In function `SerialPortLinux::setLock()':
	SerialPortLinux.cpp:(.text+0x949): undefined reference to `dev_lock'
	build/SerialPortLinux.o: In function `serialPortsScanner(std::vector<std::string, std::allocator<std::string> >&)':
	SerialPortLinux.cpp:(.text+0x194f): undefined reference to `dev_testlock'
	collect2: error: ld returned 1 exit status
	make: *** [build/SmartServoGui] Error 1

emericg, 2015-08-21:
Is that still your manual build line? The "-llockdev" is given to the compiler and not the linker.

The CMake file has been recently created to build the library while the scons was there since the begining of the project to build the various examples.

romainreignier, 2015-08-21:
Ok for the CMake and Scons.

For my issue with qmake. No, it is not my manual build line, I do the same things as the README:

    $ cd gui/
    $ qmake-qt4
    $ make

(The same with `qmake`).
I do not know the configuration of the `*.pro` files but it seems to be an issue here.

emericg, 2015-08-21:
Humm, that's weird. I'll toy around with qt4 as I see that's why you are using.

romainreignier, 2015-08-21:
I have qt4 and qt5 installed so when I run `qmake`, I have the following header in the `Makefile`:

	#############################################################################
	# Makefile for building: build/SmartServoGui
	# Generated by qmake (3.0) (Qt 5.2.1)
	# Project:  SmartServoGui.pro
	# Template: app
	# Command: /usr/lib/x86_64-linux-gnu/qt5/bin/qmake -o Makefile SmartServoGui.pro
	#############################################################################

And this one with `qmake-qt4`

	#############################################################################
	# Makefile for building: build/SmartServoGui
	# Generated by qmake (2.01a) (Qt 4.8.6) on: ven. ao�t 21 13:33:07 2015
	# Project:  SmartServoGui.pro
	# Template: app
	# Command: /usr/bin/qmake-qt4 -o Makefile SmartServoGui.pro
	#############################################################################

Then, both give me the same errors.

emericg, 2015-08-21:
Ok I tried something, can you check if it works now?

romainreignier, 2015-08-21:
Nice!
It is now working both with Qt5 and Qt4 :v: 
Thanks.
One thought, a button to turn off the torque of all the servos would be convenient.

closed 2015-08-21 by romainreignier