emericg/SmartServoFramework #3: g++-4.6 cannot compile

romainreignier, 2015-08-19:
Hi!

It seems that GCC 4.6 is not able to compile SmartServoFramework (on Ubuntu ARM 12.04):

	scons: Reading SConscript files ...                                                                                                                                                                                                           
	> LINUX platform                                                                                                                                                                                                                              
	> Gcc 4.6 compiler                                                                                                                                                                                                                            
	scons: done reading SConscript files.                                                                                                                                                                                                         
	scons: Building targets ...                                                                                                                                                                                                                   
	g++ -o examples/ex_advance_scanner.o -c -O2 -std=c++0x -pthread -Wno-unused-parameters examples/ex_advance_scanner.cpp                                                                                                                        
	In file included from examples/../src/Dynamixel.h:31:0,                                                                                                                                                                                       
		             from examples/../src/DynamixelSimpleAPI.h:26,                                                                                                                                                                                
		             from examples/ex_advance_scanner.cpp:33:                                                                                                                                                                                     
	examples/../src/ControlTables.h:58:28: sorry, unimplemented: non-static data member initializers                                                                                                                                              
	examples/../src/ControlTables.h:58:28: error: ISO C++ forbids in-class initialization of non-const static member 'reg_index'                                                                                                                  
	examples/../src/ControlTables.h:59:28: sorry, unimplemented: non-static data member initializers                                                                                                                                              
	examples/../src/ControlTables.h:59:28: error: ISO C++ forbids in-class initialization of non-const static member 'reg_addr'                                                                                                                   
	examples/../src/ControlTables.h:61:28: sorry, unimplemented: non-static data member initializers                                                                                                                                              
	examples/../src/ControlTables.h:61:28: error: ISO C++ forbids in-class initialization of non-const static member 'reg_addr_rom'                                                                                                               
	examples/../src/ControlTables.h:62:28: sorry, unimplemented: non-static data member initializers                                                                                                                                              
	examples/../src/ControlTables.h:62:28: error: ISO C++ forbids in-class initialization of non-const static member 'reg_addr_ram'                                                                                                               
	examples/../src/ControlTables.h:63:28: sorry, unimplemented: non-static data member initializers                                                                                                                                              
	examples/../src/ControlTables.h:63:28: error: ISO C++ forbids in-class initialization of non-const static member 'reg_size'                                                                                                                   
	examples/../src/ControlTables.h:64:28: sorry, unimplemented: non-static data member initializers                                                                                                                                              
	examples/../src/ControlTables.h:64:28: error: ISO C++ forbids in-class initialization of non-const static member 'reg_access_mode'                                                                                                            
	examples/../src/ControlTables.h:65:28: sorry, unimplemented: non-static data member initializers                                                                                                                                              
	examples/../src/ControlTables.h:65:28: error: ISO C++ forbids in-class initialization of non-const static member 'reg_value_def'                                                                                                              
	examples/../src/ControlTables.h:66:28: sorry, unimplemented: non-static data member initializers                                                                                                                                              
	examples/../src/ControlTables.h:66:28: error: ISO C++ forbids in-class initialization of non-const static member 'reg_value_min'                                                                                                              
	examples/../src/ControlTables.h:67:28: sorry, unimplemented: non-static data member initializers                                                                                                                                              
	examples/../src/ControlTables.h:67:28: error: ISO C++ forbids in-class initialization of non-const static member 'reg_value_max'                                                                                                              
	cc1plus: warning: unrecognized command line option "-Wno-unused-parameters" [enabled by default]                                                                                                                                              
	scons: *** [examples/ex_advance_scanner.o] Error 1                                                                                                                                                                                            
	scons: building terminated because of errors.

On the same plateform, I have tested GCC 4.7 and 4.8 and all is ok, apart from the issue #2.

emericg, 2015-08-20:
Ok this easy to fix, I'll patch it when I'm back at home.
Some people where also using (and needing) GCC 4.6 with ubuntu 12.04 back at Inria, so I'm a little surprise there is still these kind of init left in the code (must have been introduced since then).

emericg, 2015-08-20:
Should be allright now.

romainreignier, 2015-08-21:
Hi!
Thanks for the quick fix!
It is almost good but there is the `src/ServoHerculeX.h` which contain the initialization of these two data members:

    gotopos
    gotopos_commit

romainreignier, 2015-08-21:
As a side effect, I have then messages complaining about model 0 in Control Table. The values should be initialized in the constructor I think.
(With `ex_basic_test.cpp`)

	======== DEVICE SCANNING ========
	[I][ D-API][servoScan] > Scanning for Dynamixel devices on '/dev/ttyACM0'... Range is [0,253]
	.[I][ D-API][servoScan] [#1] Dynamixel servo found!
	[I][ D-API][servoScan] [#1] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#2] Dynamixel servo found!
	[I][ D-API][servoScan] [#2] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#3] Dynamixel servo found!
	[I][ D-API][servoScan] [#3] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#4] Dynamixel servo found!
	[I][ D-API][servoScan] [#4] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#5] Dynamixel servo found!
	[I][ D-API][servoScan] [#5] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#6] Dynamixel servo found!
	[I][ D-API][servoScan] [#6] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#7] Dynamixel servo found!
	[I][ D-API][servoScan] [#7] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#8] Dynamixel servo found!
	[I][ D-API][servoScan] [#8] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#9] Dynamixel servo found!
	[I][ D-API][servoScan] [#9] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#10] Dynamixel servo found!
	[I][ D-API][servoScan] [#10] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#11] Dynamixel servo found!
	[I][ D-API][servoScan] [#11] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#12] Dynamixel servo found!
	[I][ D-API][servoScan] [#12] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#13] Dynamixel servo found!
	[I][ D-API][servoScan] [#13] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#14] Dynamixel servo found!
	[I][ D-API][servoScan] [#14] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#15] Dynamixel servo found!
	[I][ D-API][servoScan] [#15] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#16] Dynamixel servo found!
	[I][ D-API][servoScan] [#16] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#17] Dynamixel servo found!
	[I][ D-API][servoScan] [#17] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'
	[I][ D-API][servoScan] [#18] Dynamixel servo found!
	[I][ D-API][servoScan] [#18] model: '18' (AX-18A)
	[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'

Another issue, but I do not understand the reason, the output lines are shift :

	======== MAIN LOOP ========

	> Use the keypad (+/-) to move (press ESC or any other key to exit)
	-Current position is: 473. Moving to 409
		                          +Current position is: 406. Moving to 470
		                                                        +Current position is: 470. Moving to 534
		                                                                                      -Current position is: 534. Moving to 470
		                                                                                                                    -Current position is: 467. Moving to 403
		                                                                                                                                         > EXIT (key was: 113)

	======== EXITING ========
	[I][SERIAL][removeLock] Lock removed for device '/dev/ttyACM0'

So I have replaces the `std::endl` by "\r\n". But it is the first time I saw this behavior in Linux.

emericg, 2015-08-21:
Lol, I was operating under the assumption that "testing is cheating"? I'll fix that ^^

emericg, 2015-08-21:
Ok it *should* be alright now, I even tested it! I'm going away for the week end, feel free to close this bug if the fix works.

romainreignier, 2015-08-21:
Ok, the building is ok. But I still have the following line for each servo discovered:

    [E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_model: '0'

But it is not related to gcc-4.6 anymore.

closed 2015-08-21 by romainreignier