emericg/SmartServoFramework #4: Documentation and dependencies

romainreignier, 2015-08-19:
It would be nice to give some instructions for the building process of the library and examples.
I did not know `scons` so I had to look for it.
Now I know that it is a simple line:

    $ scons .

But as there are some instructions for the GUI, why not for the library?

Also for the examples, a `SConstruct` or better `CMakeLists.txt` would be nice.

Another point is the dependencies. I had to install `liblockdev1-dev`, `scons` and even `doxygen`. I understand that Doxygen is a nice tool and you made a great job for documenting the code, but maybe it is not useful to make it as a dependence for the `CMakeList.txt` as it is not used in the building process (I think).

So for the dependencies, CMake does the job to remind us to install it, but a line about that in the README would be nice too :)

I understand that it is not so simple for a multi-platform project and those things are just details. But I think they can improve the first experience of your nice tool ;)

emericg, 2015-08-20:
Hi Romain!
It's very nice to have someone reporting issues and taking a look at the software, thanks!

I recently started working again on the framework and it's software, I want to publish a "1.0" with:
- A working and tested GUI on linux, mac and windows. People needs to be able to just download an archive an launch the software!
- An effort on the SimpleAPI and its documentation (the ControllerAPI probably need works and some more design thoughts). 1.0 should also provide a stable API, so I can start breaking everything in new branchs ;-)
- And yes, more docs!

- Regarding the build systems, I don't know if I want to keep the scons method at all. CMake will do a better job I think.
- Doxygen, is definitly not needed, the CMakeLists.txt just tells you it can be installed. I wanted to use this to generate the API documentation automatically if it's installed.
- The lockdev dependency is not ideal, I think I'll defaut to another method.

So yes, I'll update the readme's in a few days, and I also want to work on a github wiki ;-)
Thank you again for the inputs!

emericg, 2016-03-13:
OK I'll close this issue now 'better late than never).
Right now we have way better documentation about building and using the framework, improved build systems and better Mac OS support.

closed 2016-03-13 by emericg

romainreignier, 2016-03-14:
:+1: 
I do not use the Dynamixel for now so I did not noticed the recent changes but it is nice to see this nice tool evolving and I will surely use it the next time I have to use Dynamixels.