emericg/SmartServoFramework #6: CMakeLists.txt file missing in SmartServoFramework v0.9

pauljurczak, 2015-10-04:
I just started playing with SmartServoFramework and I noticed that  CMakeLists.txt file is missing from v0.9 download. It is present in master branch, which seems to install OK.

BTW, thank you for all the effort you put into this library. This is just what I was looking for - perfect for a quick prototype with Herkulex servos.

emericg, 2015-10-04:
Hi!
Do you mean the CMakeLists.txt file is missing from the archive we can download on the "Download ZIP" button on the main page? That's weird, I just tried it and it was there as expected. Can you tell me a bit more about your issue?

And your welcome ;-) Do not hesitate to post your finding, bugs, and remarks. I'm not working full time on this anymore, so having a little feedback keeps me motivated!
And the HerkuleX code is not as tested as the Dynamixel code, so a little bugfix can only be good...

pauljurczak, 2015-10-04:
Hi,

No, CMakeLists.txt is present in v0.93 linked from the "Download ZIP" button on the main page, but it is not present in v0.9 linked from the "release" button near the top of the main page.

Thank you, for responding so quickly.

emericg, 2015-10-05:
Ho damn I didn't even remember doing a 0.90 release!
Well until recently the framework didn't use CMake at all, that's why this file isn't in the 0.90 archive.

I'll update the readmes and documentation, fixe your issues with MSVC so Windows support can be on par with Mac and Linux support, and will release a 0.95 aka "beta 1".

closed 2015-10-05 by emericg