emericg/SmartServoFramework #7: No .lib file built with v0.93 on Windows

pauljurczak, 2015-10-04:
I installed current master branch v0.93 on Windows 7 64-bit with VS2013 and I've built all projects. SmartServoFramework.dll file was generated, but there is no corresponding SmartServoFramework.lib file. Did I miss something?

emericg, 2015-10-04:
Hum, unfortunately I have never tried doing that... I am not a windows user at all, but from what I understand you cannot link with the DLL without this .lib file right?
Did you use the CMake feature that generate project file for MSVC? Could you walk me trough the procedure, it would be cool to have that put into the readme!

When testing the framework on windows, I was just putting all the files into my projects. Now that we can use it as a library it's way more convenient.
But of course it doesn't work on windows as on the other OS's... I'll have a look at this page this afternoon: https://cmake.org/Wiki/BuildingWinDLL

Also I noticed that the serial port auto-detection doesn't work when building with MSVC instead of mingw, be carefull with that! The string handling is working a bit differently and I didn't took time to fix it yet.

Vaufreyd, 2015-10-04:
Hello,

I think that there is not export (specific to Windows as on Windows, unlike on Linux you select which symbol to export on your library - it is not the same paradigme as the static function also supported on Windows). If there is not symbol to export, no lib is mandatory.

Try to do a static lib.

Best regards. Doms.

PS: Emericcg, what do you mean by auto-detection is not working ?

pauljurczak, 2015-10-04:
@emericg You are right, you need .lib file to build the executable. I'm using Visual Studio projects generated by CMake. This is not a showstopper for me, because .lib file can be generated from .dll file, which builds fine. I don't have enough knowledge about VS build system to recommend changes in project files.

As for serial port autodetection, it worked fine with the test program I've just built.

pauljurczak, 2015-10-04:
@Vaufreyd That's what I did. I've built a static library by manually editing VS project file.

emericg, 2015-10-05:
OK I will specify a list of symbols to export then.

By auto-detection not working I mean serial port auto-detection. When connecting to a device you only need to set "auto" instead of it's full path like "//COM8".
From what I can recall, the serialPortsScanner() function from SerialPortWindows.cpp doesn't output any results when builded with MSVC, thus breaking the auto-detection. But it's good news if you confirm it's actually working!
On Windows, the auto-detection process lists the /COMxx ports and then try to connect to the last one found (as the first ones are *usually* some stuff put by the system, and if you try to use these well you end up on a blocking open() call). I'm not sure how to handle that better.

@Vaufreyd It's nice to see that you are still keeping an eye on the framework's development ;-)

pauljurczak, 2015-10-06:
@emericg Both "auto", "\\COM4" and "\\\\.\\COM4" worked in my case with `Serial interface successfully opened` message.

emericg, 2019-01-17:
I guess better late than never, the .lib file should now be created by default as long as cmake 3.4+ is used!

closed 2019-01-17 by emericg