emericg/SmartServoFramework #8: Visual Studio 2013 build issues

pauljurczak, 2015-10-04:
I'm using the current master branch v0.93, CMake to generate VS project files and Visual Studio 2013 Update 4 on Windows 7 64-bit. I noticed the following issues:

1. I'm getting: 

`Project: SmartServoFramework_shared, Configuration: Release x64 ------`
`Command line error D8021: invalid numeric argument '/Wno-unused'`
`Command line error D8021: invalid numeric argument '/Wno-unused-parameter'`

which I fixed by manually editing SmartServoFramework_shared project file.

2. When I build the whole SmartServoFramework solution, INSTALL project doesn't produce `include` directory. I have to rerun INSTALL project separately in order to get `include` directory.

emericg, 2015-10-05:
Your n°1 issue should only produce warning right? Did you solved it by just removing " -Wno-unused -Wno-unused-parameter" from the "set(CMAKE_CXX_FLAGS ...)" line? If so I'll remove those but I suspect other options should be causing problems with MSVC.

The INSTALL option is not supposed to work with Windows so I guess that's normal. I really don't know where am I supposed to put these includes anyway, so feedback are welcome :-)

pauljurczak, 2015-10-06:
@emericg No, issue #1 produces an error and fails the build. I edited the project properties inside Visual Studio IDE and removed `-Wno-unused -Wno-unused-parameter`. I didn't notice any problem caused by that, but I still didn't get my HerkuleX servos to move. I will get back to it in a few days.

emericg, 2015-10-15:
By the way this should be fixed by the last commit.
Regarding HerkuleX servos not moving, from what I experienced with these device, you need to activate the torque manually each time they are powered on. Keep us posted on your endeavours!

closed 2015-10-15 by emericg