emericg/SmartServoFramework #9: Serial port not working on El Capitan (Mac OSX 10.11)

srandoux, 2015-11-10:
Hi,

I upgraded to macosx 10.11 and after doing some tests the tcsetattr to set the serial port with a USB2AX is not working anymore... 

I started ot look into it but for the moment i am clueless.

egrange-gpsw, 2015-11-10:
Damn, I'll try to have a look too then. Thanks for the report!

emericg, 2015-12-08:
Ok that's weird, I got it working seamlessly on one Mac, and failing with the same error you got on another one ?! Tried sudo, disabling SIP, what now?

srandoux, 2015-12-08:
I tried on 3 macs, and same error on all for them... I guess for the moment it is better to stick to 10.10. (As far I understand at least). I first thought it was related to the new mandated security model also but it does not seem to be the case. I tried to use the USB2AX in a vmware fusion 7.0 virtual machine running windows 7 with dynamixel robot plus and it was not recognized anymore. I upgraded to vmware fusion 8.0 and it works again. I guess apple made some changes in the USB stack which are not handled properly by the dongle and thus some system calls fails....
I downloaded the source code of the firmware and checkedif  it was easy to debug it on the USB2AX but it can not be done in 5 minutes.... Maybe we should discuss with the guy who wrote the firmware, it is more an OSX issue, and today, USB2AX can not be used at all with MacOSX 10.11.
Do you want me to get in touch with USB2AX people?At least we can try...

emericg, 2015-12-08:
Ho you are gonna like this, I set a sleep(1) this time just before the tcsetattr(), around line 490... Try this and keep me posted!

srandoux, 2015-12-08:
ok, I'll do. I guess it was one of the things I try but it works that is fair enough....

srandoux, 2015-12-08:
No luck.... 1 or several sleeps... will try another mac...

>> scanServos(/dev/cu.usbmodem1421)
[I][ C-API][updateInternalSettings] - Using Dynamixel communication protocol version 1
[I][SERIAL][SerialPortMacOS] - Device name has been set to: 'cu.usbmodem1421'
[I][SERIAL][SerialPortMacOS] - Device node has been set to: '/dev/cu.usbmodem1421'
[I][SERIAL][SerialPortMacOS] - Device baud rate has been set to: '1000000'
[I][SERIAL][setLock] - TIOCEXCL lock set for: '/tmp/cu.usbmodem1421.lock'
[E][SERIAL][openLink] Error setting tty attributes /dev/cu.usbmodem1421 - Invalid argument(22).
Error, cannot start scannig for servos! (scann running ? 1)   (controller ? 0x0)
[I][ C-API][updateInternalSettings] - Using Dynamixel communication protocol version 1
[I][SERIAL][SerialPortMacOS] - Device name has been set to: 'cu.usbmodem1421'
[I][SERIAL][SerialPortMacOS] - Device node has been set to: '/dev/cu.usbmodem1421'
[I][SERIAL][SerialPortMacOS] - Device baud rate has been set to: '57600'
[I][SERIAL][setLock] - TIOCEXCL lock set for: '/tmp/cu.usbmodem1421.lock'
[E][SERIAL][openLink] Error setting tty attributes /dev/cu.usbmodem1421 - Invalid argument(22).
Error, cannot start scannig for servos! (scann running ? 1)   (controller ? 0x0)
[I][ C-API][updateInternalSettings] - Using Dynamixel communication protocol version 2
[I][SERIAL][SerialPortMacOS] - Device name has been set to: 'cu.usbmodem1421'
[I][SERIAL][SerialPortMacOS] - Device node has been set to: '/dev/cu.usbmodem1421'
[I][SERIAL][SerialPortMacOS] - Device baud rate has been set to: '1000000'
[I][SERIAL][setLock] - TIOCEXCL lock set for: '/tmp/cu.usbmodem1421.lock'
[E][SERIAL][openLink] Error setting tty attributes /dev/cu.usbmodem1421 - Invalid argument(22).
Error, cannot start scannig for servos! (scann running ? 1)   (controller ? 0x0)
[I][ C-API][updateInternalSettings] - Using HerkuleX communication protocol
[I][SERIAL][SerialPortMacOS] - Device name has been set to: 'cu.usbmodem1421'
[I][SERIAL][SerialPortMacOS] - Device node has been set to: '/dev/cu.usbmodem1421'
[I][SERIAL][SerialPortMacOS] - Device baud rate has been set to: '115200'
[I][SERIAL][setLock] - TIOCEXCL lock set for: '/tmp/cu.usbmodem1421.lock'
[E][SERIAL][openLink] Error setting tty attributes /dev/cu.usbmodem1421 - Invalid argument(22).

srandoux, 2015-12-08:
Same on second mac, not working :-(

emericg, 2015-12-09:
Yes this was indeed the same error as this one : [E][SERIAL][openLink] Error setting tty attributes /dev/cu.usbmodem1421 - Invalid argument(22).

Sorry I really thought the timer would do the trick (again ^^).
I did try two other things on the faulty mac, first, disabling the lock on the /dev/cu.usbmodem, second, removing the "usb2ax" from the network adapter.

orgicus, 2016-08-11:
Hi,

I seem to have a somewhat similar issue. On OSX 10.11 SmartServoGui using Herkulex servos doesn't seem to see the FTDI Serial converter I'm using. On Windows the same converter works fine with the Herkulex Manager software.

Console output looks like this:
```
./SmartServoGui 
Warning: no configuration file, using default values! A new one will be created next time you change a setting.
Parsing settings from fallback string to document succeeded.
[I][SERIAL][serialPortsScanner] serialPortsScanner() [MacOS variant]
[E][SERIAL][serialPortsScanner] Could not get path for modem.
[I][SERIAL][serialPortsScanner] serialPortsScanner() [MacOS variant]
[E][SERIAL][serialPortsScanner] Could not get path for modem.
[I][SERIAL][serialPortsScanner] serialPortsScanner() [MacOS variant]
[E][SERIAL][serialPortsScanner] Could not get path for modem.
saveSettings() 
```

I've tried to set the path to the serial port in settings: ```/dev/tty.usbserial-FTHBYOES``` but I'm not sure the settings are saved. I've check and ```/Library/Preferences/SmartServoGUI.json``` does not exist.

I've tried some pretty messy things like changing the ```MATCH_PATH``` in SerialPortMacOS to ```/dev/tty.usbserial-FTHBYOES```. Similarly I've changed the enabled the serial port and hardcoded the path and baud rate in ```fallback_json``` in settings.cpp as well, but no change. 
What's strange is that I didn't see these tweaks reflect in the Settings panel either.

How can I enable SmartServoGui to see and open the serial converter I'm using ?

Thank you,
George

srandoux, 2017-02-11:
It works again with  Macosx 10.12! you can close the issue

closed 2017-02-11 by srandoux