emericg/SmartServoFramework #10: Debian Serial Port Errors

MiddleMan5, 2016-03-02:
I'm getting random timeout errors and value errors, the error output keeps getting tabbed over making the messages nearly unreadable. All (6) servos have the same (10) timeout value and are each being given proper setPosition values.

If I could get the location in the code where these are generated I might be able to work up a patch.

`[E][ SERVO][updateValue] [#31] updateValue(reg 31 / CURRENT_POSITION to -1) [REGISTER VALUE ERROR] (min: 0 / max: 65536)
                                                                                                                         [E][   DXL][dxl_print_error] [#31] COMM_RXTIMEOUT: Timeout reached while waiting for a status packet!
                              [E][ SERVO][updateValue] [#40] updateValue(reg 32 / CURRENT_SPEED to -1) [REGISTER VALUE ERROR] (min: 0 / max: 65536)
                                                                                                                                                   [E][   DXL][dxl_print_error] [#40] COMM_RXTIMEOUT: Timeout reached while waiting for a status packet!
                                                        [E][ SERVO][updateValue] [#40] updateValue(reg 34 / CURRENT_LOAD to -1) [REGISTER VALUE ERROR] (min: 0 / max: 65536)
                                                                                                                                                                            [E][   DXL][dxl_print_error] [#40] COMM_RXTIMEOUT: Timeout reached while waiting for a status packet!
                                                                                 [E][ SERVO][updateValue] [#40] updateValue(reg 32 / CURRENT_SPEED to -1) [REGISTER VALUE ERROR] (min: 0 / max: 65536)
      [E][   DXL][dxl_print_error] [#40] COMM_RXTIMEOUT: Timeout reached while waiting for a status packet!
                                                                                                           [E][ SERVO][updateValue] [#40] updateValue(reg 34 / CURRENT_LOAD to -1) [REGISTER VALUE ERROR] (min: 0 / max: 65536)
                               [E][   DXL][dxl_print_error] [#40] COMM_RXTIMEOUT: Timeout reached while waiting for a status packet!
                                                                                                                                    [E][ SERVO][updateValue] [#40] updateValue(reg 39 / MOVING to -1) [REGISTER VALUE ERROR] (min: 0 / max: 256)
                                                [E][   DXL][dxl_print_error] [#40] COMM_RXTIMEOUT: Timeout reached while waiting for a status packet!
                                                                                                                                                     [E][ SERVO][updateValue] [#40] updateValue(reg 31 / CURRENT_POSITION to -1) [REGISTER VALUE ERROR] (min: 0 / max: 65536)
                                                                             [E][   DXL][dxl_print_error] [#40] COMM_RXTIMEOUT: Timeout reached while waiting for a status packet!
[E][ SERVO][updateValue] [#42] updateValue(reg 31 / CURRENT_POSITION to -1) [REGISTER VALUE ERROR] (min: 0 / max: 65536)
                                                                                                                        [E][   DXL][dxl_print_error] [#42] COMM_RXCORRUPT: Status packet is corrupted!
`

MiddleMan5, 2016-03-02:
I just wanted to comment on the level of genius I believe emericg to be.
Traces actually gives the exact file that is throwing the errors.
I went into Servo.cpp and Dynamixel.cpp and replaced all "\n" with "\r\n" for good measure.
Sure enough all of the formatting is coming out properly now.

The errors are probably me trying to write and read from the servos too fast.

emericg, 2016-03-04:
Hey !
It's nice that you found MiniTrace usefull :-) There is quite a few features that can be easily toggled in **minitraces_conf.h**, like timestamps, file/function/line that outputted the trace, colors, flush after each trace (and other stuff hidden). And obviously tracing level for each submodules.

Your first problem (the endlines) is a bit weird. Usually we only use '\n' under unixes OS to change lines. The '\r\n' is typically Windows specific behaviour, but thinking back I remember using this framework under Windows without any new line problem, so I don't really known what to tell you.
The new line could certainly be embedded inside the TRACE_xxx macros, or we could also juste add a macro after the text (but before the argument) to abstract these caracters and making that easily configurable.
Can I get a little more info on your system and terminal emulator ?

Regarding your errors, first what timeout value are you refering to? The per servo "Return Delay Time" or the latency timer per serial port? or even an other one that I forgot about?
Timeout problems are very common with Dynamixel (and probably most) serial devices.
I trust that you read the "serial port" section of the readme?

To dig a little deeper into the issue, the thing is, serial commands are usually slow, and latency over the bus (or most of the time over the USB to TTL converter) are degrading performance even more. A **lot** of things can be tweaked, and we can even choose to wait for the devices to return every values instead of issuing timeouts, but from experience:
- latency can come from a mix of serial port / USB adapters / servos / commands / and probably how I handle the serial port ^^
- some Linux systems are handling the FTDI latency tweaks better than other.
- some servos models can have very different response time (depending the CPU / running frequency used)
- some particular registers can (sometimes) takes WAY more time to return values than others. For instance I think I remember MX servos and max torque values taking more than 100ms to return values. So do we want to wait for these (wich would makes us use insame timeout timers) or do we just ignore them?

Your particular errors comes from the fact that a servo didn't return a queried value on time. These are not catastrophic errors (like failing to set a new position) but are still annoying I'll give you that. These values are being read by your Controller API, which maintains a set of "fresh" values for the most commonly used registers.
Currently the code communicating with the servos write the value 0 (or maybe -1) when a timeout occurs, and the code wich queried this register value interpreted the value 0 as corrupted. Because this is multithreaded code and I was probably lazy, there is no other way for these two threads to get the timeout error without corrupting the last good value for a register. Depending on the case we actually do not want to use an old value. I think a bit of refactoring would be needed to handle these timeout more cleanly. But first there is probably a way to be just less aggressive when printing these errors ;-)

MiddleMan5, 2016-03-05:
I've only recently started delving into programming at a library and system level, prior to this the only experience I had were on hobby MCU boards like the Arduinos as well as Dev Kits from Atmel. So Excuse me for my naivety in this subject.

A cat of my /proc/version file lists:
`Linux version 3.8.13-bone74  (gcc version 4.9.2 (Debian 4.9.2-10) ) `
The architecture specifically runs on a BeagleBone Black Rev. C.
The Serial Transceiver is a USB2AX module.
Currently my system controls 6 AX-12A's, with plans to add at least 7, if not 12 more.

Originally I was using Netbeans as a cross compilation IDE, but when I tried to merge in your code (after racking my brains on building my own) Netbeans blew up. Right now I'm just using the built in text editor in my windows term program "Mobaxterm 8.0" with a modified Scons script that compiles my robots source code and libraries in together with your source files. It's not pretty but it works.

Now I just realized that Moba has given me similar line-feed issues before, but I think it might be a mix of your code AND my terminal. I never thought to check my settings. I notice that if the software ever crashes in the middle of an operation, my terminal will stop sending carriage returns all-together. I believe this might be due to the call to "system("/bin/stty raw");" without a crash method that restores the "Cooked" mode. Using my Ubuntu Laptop's build in ssh, and my android Phone's "Juice SSH" terminal clients also produced these line-feed errors, but I ended up fixing them with the aforementioned "\r\n" approach. 

Honestly your software is everything I was looking for and more; from my point of view it is truly amazing. I'm an EE student and I'm working on building a quadruped following inspiration from labs like Boston Dynamics. I wanted to build something more than just your average hobby mech. I wanted to be able to poll each joint for various information, and be able to tune the operation based on the feedback. I chose the BeaglBone because of it's small size and relative robustness. Originally I was using hobby servos, but the lack of a networking system made it a total mess to route cables. I ended up settling (for lack of a better option) on the rather bulky and expensive AX-12A's. 

Right now I have the rear half of the robot functional and working, but it's control code is an absolute mess. I'm not much of a programmer. I'm looking to build an interface system that provides valuable information about the operating statistics of the servos. I'm sure your framework would do the trick, but it's a little large and supports more options than I need it, and I found the auto-generated Doxygen files to be somewhat broken and lacking information. Understandably you wrote it for yourself, but I'd love to be able to get to a point of understanding that I might optimize it and fix some of the "laziness" as you say.

One of my bigger problems is that the error messages seemed to stop after I increased the servo's Return Delay Time register, but this still isn't perfect, and the longer the program runs, the more frequent the error messages. The errors also cause a momentary pause in servo movement, which ends up making the system shake and jerk. 

Any support you can give would be greatly appreciated, and please ask if I've left any information out. With a little guidance I would also be interested in helping write support documentation and such if you need an extra hand. For my work, this framework is truly very useful.

MiddleMan5, 2016-03-06:
Okay, I've been fiddling with the code again today.

One little nitpick is that waitMovmentCompletetion is misspelled in a couple files, I created a pull request if you want to look at it.

Now the servos seem to be acting really finicky, setMovingSpeed doesn't seem to do anything except put random values into the speed register. I haven't actually looked at the code yet.
Sometimes (very rarely) the speed register changes to a value between 40-100, but most times it does absolutely nothing.

I'm also having trouble getting the torque (among many registers) to turn on and off reliably, when I try to change the torque register to '0' at program exit (right before the controller is disconnected) the Servo's fail to release torque. Actually this gives me an idea, it seems as though segments of the code will execute and "muffle" other parts of the code, IE: 'TURN OFF TO...CTRL EXI...END PROGRAM' (if you get what I mean).

emericg, 2016-03-06:
Hi,
I've been doing something else today, but I'll be working on SmartServoFramework this week (woohoo !). Mainly documentation and a few items to stabilize current version.

Just a quick reply, I assume you are using waitMovmentCompletetion() since you mention it. It is known to cause problems, so maybe don't use it until we can see what's wrong ^^
You are also using the ControllerAPI, which is pretty nice but require knowledge on how the framework and the servos works. It will be good to talk about it, maybe we will be able to crank up the documentation on that!

MiddleMan5, 2016-03-06:
No rush, thanks for taking the time to check back in. I'm removing the waitMovmentCompletion from my calls, looking forward to corresponding on this project with you!

emericg, 2016-03-12:
Replacing the '\n' by '\r\n' really doesn't seems like a proper solution, only a hack fighting another hack.

The "system("/bin/stty raw");" has the potential to wreak havoc in any terminal, that's true. Unless you got a really good reason to use it, you should remove it (the example code is using it to get keyboard input without pressing enter each time, so it seems related to your issue).

emericg, 2016-03-13:
Ok with latest commits you should have only one error trace printed (the COMM_RXTIMEOUT, not the redondant one from updateValue())

closed 2017-11-18 by emericg