emericg/SmartServoFramework #12: Lack of Multi Turn mode

romainreignier, 2016-04-22:
Hi Emeric,
It seems that the Multi-Turn mode of the Dynamixel MX servos is not supported by SmartServoGui.
A radio button would be nice. But even when setting CW angle limit and CCW angle limit to 4095, we cannot put a position higher than 4095, the value is rejected by the software.
I think it would be a nice feature to support but maybe not so widely used.
Romain

emericg, 2016-08-20:
Is multi turn mode a new feature of a recent firmware? I remember the HerkuleX device having 900° of rotation but not the Dynamixel. Unfortunatly I don't have any MX servos to finish this so I'll have to improvise.

romainreignier, 2016-08-20:
I have no idea how old is that feature.
I only know the Dynamixel family for 1,5 year.
But I have also noticed that the library [dynamixel_h](https://github.com/HumaRobotics/dynamixel_hr) lacks this feature too.
At work, I should be able to use some MX-28AT motors to do some testing if needed.