emericg/SmartServoFramework #13: Can't open servo connection on El Capitan (Mac OSX 10.11)

orgicus, 2016-08-12:
Hi,

I tried to use the library in a minimal c++ setup, but it seems to hang when connecting.
Here's what I've tried so far based on the simple example:
```c++
HerkuleXSimpleAPI hlx(SERVO_DRS_0101);
    std::string deviceName = "/dev/tty.usbserial-FTHBYOES";
    hlx.connect(deviceName, 115200,SERIAL_OTHER_FTDI);
//    if (hlx.connect(deviceName, 115200) == 0)
//    {
//        std::cerr << "> Failed to open a serial link for our SimpleAPI! Exiting..." << std::endl;
//    }
    
    std::vector <int> ids = hlx.servoScan();
    std::map <int, int> mCurrentPos;
    std::map <int, int> mWantedPos;
    
    // Read initial position(s) for each servo found by the scan
    std::cout << std::endl << "Servos: [";
    for (std::vector <int>::iterator idsIterator = ids.begin(); idsIterator != ids.end(); idsIterator++)
    {
        // Read & store position
        int pos = hlx.readCurrentPosition(*idsIterator);
        mCurrentPos.insert(std::pair<int,int>(*idsIterator, pos));
        mWantedPos.insert(std::pair<int,int>(*idsIterator, pos));
        
        // Print id/position pair for current servo
        std::cout << "id: " << *idsIterator << " (pos:" << mCurrentPos.at(*idsIterator) << "), " ;
    }
    std::cout << "]" << std::endl;
```

If I simply run it, the code hangs on connect.
If I add a breakpoint on hlx.connect I get more info:
```
[[1;32mI[0m][ D-API][1;37m[HerkuleXSimpleAPI][0m - Using HerkuleX communication protocol
[[1;32mI[0m][SERIAL][1;37m[SerialPortMacOS][0m - Device name has been set to: 'tty.usbserial-FTHBYOES'
[[1;32mI[0m][SERIAL][1;37m[SerialPortMacOS][0m - Device node has been set to: '/dev/tty.usbserial-FTHBYOES'
[[1;32mI[0m][SERIAL][1;37m[SerialPortMacOS][0m - Device baud rate has been set to: '115200'
[[1;31mE[0m][SERIAL][1;37m[openLink][0m Unable to open device on serial port: '/dev/tty.usbserial-FTHBYOES', Interrupted system call(4)
[[1;32mI[0m][ D-API][1;37m[servoScan][0m > Scanning for HerkuleX devices on '/dev/tty.usbserial-FTHBYOES'... Range is [0,253]
[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[... 463 lines cut ...]
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.[[1;31mE[0m][SERIAL][1;37m[tx][0m Cannot write to serial port '/dev/tty.usbserial-FTHBYOES': invalid device!
[[1;31mE[0m][   HKX][1;37m[hkx_txrx_packet][0m Unable to send TX packet on serial link: '/dev/tty.usbserial-FTHBYOES'
.

Servos: []
Aug 12 11:19:55  HerkuleXTestDebug[758] <Warning>: void CGSUpdateManager::log() const: conn 0xc55b: spurious update.
```

Why can't SmartServo's serial port open/write ? I can write to the serial port with ```screen```/CoolTerm/etc.  just fine.

Is there a way to use SmartServo in conjuction with a different Serial class ?

emericg, 2016-08-14:
Hi,
I'm not really sure what's happening, but first can you try with root permissions?
It is possible to use the framework with a different serial classe, if you write it of course ^^ The macOS implementation as always been a little shaky, I was planning to write a new, os independant class using "Qt Serial Port", that may be worth the effort.
Keep me posted on your issue, we'll find something.

emericg, 2016-08-14:
Also, can you try using the default SERIAL_UNKNOWN in the  hlx.connect() call?

orgicus, 2016-08-14:
Hi,

I've tried with root persmisions and also tried SERIAL_UNKNOWN, but the output looks exactly the same. I'll try to use a different serial port OSX implementation. In terms of SmartServoFramework, as far as I can tell, I need to refactor the Herkulex classes to allow different external serial port classes, other than SerialPort/SerialPortMACOS. Is my assumption correct or is there a quick/easy way to get the bytes straight after a servo instruction, just before sending to the serial port ?

Thank you,
George

emericg, 2016-08-20:
Hi, I'm sorry I'm responding this late.
In order to implement a new serial port implementation, you juste need to create a new class that inherit the base SerialPort class found in SerialPort.h, there is no need for other modifications inside the framework. Just like the windows/linux/mac variant are doing. You can also start by just copy/pasting the mac variant and rewriting it's "insides".

orgicus, 2016-11-01:
Hi,

Finally had a chance to play with SmartServoFramework again.

As a quick test I did a very hacky thing: just used Qt's [QSerialPortInfo](http://doc.qt.io/qt-5/qserialportinfo.html) functionality using mostly their sample code:

```c++
int serialPortsScanner(std::vector <std::string> &availableSerialPorts)
{
    int retcode = 0;

    TRACE_INFO(SERIAL, "serialPortsScanner() [MacOS variant]\n");
    //QT test
    //http://doc.qt.io/qt-5/qtserialport-index.html
    //http://doc.qt.io/qt-5/qtserialport-examples.html
    const auto serialPortInfos = QSerialPortInfo::availablePorts();

    std::cout << "Total number of ports available: " << serialPortInfos.count() << std::endl;

    std::string description;
    std::string manufacturer;
    std::string serialNumber;

    for (const QSerialPortInfo &serialPortInfo : serialPortInfos) {
        description = serialPortInfo.description().toStdString();
        manufacturer = serialPortInfo.manufacturer().toStdString();
        serialNumber = serialPortInfo.serialNumber().toStdString();
        std::cout << description << std::endl;
        std::cout << std::endl
            << "Port: " << serialPortInfo.portName().toStdString() << std::endl
            << "Location: " << serialPortInfo.systemLocation().toStdString() << std::endl
            << "Description: " << description << std::endl
            << "Manufacturer: " << manufacturer << std::endl
            << "Serial number: " << serialNumber << std::endl
            << "Busy: " << (serialPortInfo.isBusy() ? "Yes" : "No") << std::endl;

        retcode++;

        // Add it to the list
        std::string portPath = serialPortInfo.systemLocation().toStdString();
        availableSerialPorts.push_back(portPath);
    }
    return retcode;
}
```

This worked! I could even see a few Herkulex servos attached.
However I was greeted by this message when I tried to use **Advanced Controls** and **Motor Tester**:

> This functionality will be available soon!

@emericg Any pointers on how I could advance with these two features ?

Thank you,
George

emericg, 2016-11-02:
Hey.
It's pretty nice you made this QSerialPortInfo hack!
I actually made a full QSerialPort implementation shortly after we discussed it, but got stuck on a Qt threading issue and never commited it. The new class is only working for the simple API right now.

Regarding the advance controls and motor tester:
- **advance controls** was only meant to be a more "visual" configuration tool, relying more on gui elements than the register table to configure your servos (like this screenshot: http://www.vstone.co.jp/products/dongbu_robot/img/Herkulex_manager_2.JPG)
- **motor tester** was meant to be a simple tool, making your servo move following a sin or cos function an watching the results on a graph (planned movement vs mesured movement). There is a preliminary version (ex_sinus_control.cpp) inside the examples folder. You need to enable opencv and maybe port a bit of code from dynamixel to herkulex, but it should only take a minute. (also, a bit like this screenshot: http://www.vstone.co.jp/products/dongbu_robot/img/Herkulex_manager_3.JPG)

Both these tools are probably easy to write, and if that's something that would be interesting to you I'll definitly help you with this. I lost interest a long time ago myself, but seing people using the code is always the most powerful motivator a developer can get ;-)

orgicus, 2016-11-03:
Hi,

I don't have a lot of experience with c++ and Qt, but I notice the Qt serial library command line examples are great (Command Line Reader Async,Command Line Reader Sync,Command Line Writer Async,Command Line Writer Sync).

A full QSerialPort implementation sounds nice: one class to rule them all :P
If the QSerialPort Async examples don't help, maybe keep your own threading model and try the QSerialPort Sync example approach ?

Regarding the Advance Controls and Motor tester, I was actually aiming to start with the simple things first. In your screenshot, there are dials and sliders to control the servo:

[image]

When I used the app, I found the DRS-0101 servos attached, but the UI only shows two rows on the left of the view and general info section on the right is disabled. Why would that happen ?
Any recommendations on how I could the controls for each servo ?

emericg, 2017-02-15:
By the way another user reported that things just started working again with macOS 10.12, do you think you could give it a try?

orgicus, 2017-02-15:
Hi Emeric,

Unfortunately I'm still a little paranoid about all the libraries I have installed (opencv 3.x,PCL,etc.) mishevaving when I upgrade to 10.12. I will be able to access a colleague's computer that has already been upgraded to 10.12 in a few days and I will test then.

Thank you very much for the update on that.

emericg, 2017-02-15:
I understand that very well, these updates may be easy for regular users but alway proves challenging for us developpers...
Trying on another computer seems like a good idea ;-)

emericg, 2017-04-24:
I'm closing this issue because it *seems* to be fixed by recent macOS versions... Feel free to reopen if the problem reappear.

closed 2017-04-24 by emericg