emericg/SmartServoFramework #14: Can't compile on Ubuntu Xenial 16.04 LTS

orgicus, 2017-02-15:
Hi,

I've tried to use the SmartServo UI on Linux, but I've ran into a compilation error:
```bash
SerialPortLinux.cpp:36:21: fatal error: lockdev.h: No such file or directory
compilation terminated.
```

I checked out the Build Instructions and Serial communication readmes and spotted these notes:

> liblockdev (used to lock the serial port to a single software instance, which will save you from a lot of potential head scratching errors...)
> Linux: "lockfile" support
> Linux: "liblockdev" support (default)
> Linux: "flock()" support

I've tried install any related package I could find like `liblockfile-dev`,`liblockfile-bin`,`libghc-filelock-dev`,`coreutils`,`flock` and making no difference to the missing headerfile compile error.

However I haven't figured out how to install liblockdev.

Any hints/tips on how I should proceed ?

Thank you,
George

emericg, 2017-02-15:
Hi!
Thanks for the report. It seems that lockdev has just been... removed from ubuntu! I never really got flock working but it does seems to be the better alternative to lockdev. I'll give it another go tommorow.

In the mean time you should be able to install lockdev from the package though. Go there http://packages.ubuntu.com/source/trusty/lockdev and install "liblockdev1" and "liblockdev1-dev" for your architecture. I just tryed it in a VM and it doesn't even complained about the old package.

orgicus, 2017-02-15:
Hi Emeric,

Thank you for the swift reply and taking your time to test.
I will try installing "liblockdev1" and "liblockdev1-dev".

emericg, 2017-02-20:
In the meantime I disabled the lockdev feature for the linux builds. Please let me know if that solves your problem!

emericg, 2017-04-29:
No response in a while, considering this fixed! Feel free to reopen in case of further issues.

closed 2017-04-29 by emericg