emericg/SmartServoFramework #15: Very high write time to servos

embeye, 2017-04-09:
Hi,
First of all thanks for having this library available for free. 
I am using two MX-12W servos which are supposed to have concurrent motion.I am currently using SimpleAPI which takes around 0.193 seconds to finish write to both servos.

`std::string deviceName = "/dev/ttyUSB0";

    if (motor->connect(deviceName, 1) == 0)
    {
        std::cerr << "> Failed to open a serial link for our SimpleAPI! Exiting..." << std::endl;
        exit(EXIT_FAILURE);
    }

    while(true){

        std::cout<<"Enter the pan position :";
        std::cin>>pan_pos;

        std::cout<<std::endl;

        std::cout<<"Enter the Tilt position :";
        std::cin>>tilt_pos;

        std::cout<<std::endl;

        timestamp_t t0 = get_timestamp();
        motor->setGoalPosition(PAN_MOTOR_ID,pan_pos,_speed);
        motor->setGoalPosition(TILT_MOTOR_ID,tilt_pos,_speed);
        timestamp_t t1 = get_timestamp();

        double secs = (t1 - t0) / 1000000.0L;
        std::cout<<"Time: "<<secs<<std::endl;`  
This is my piece of code for doing the work. Is it possible to get some guidance to solve the issue?
Thanks

emericg, 2017-04-22:
Hi,
I'm sorry to respond this late but I didn't immediatly saw something wrong about your code then forget to respond ^^
The thing is there isn't an actual mistake in your code that explain the slow connexion. When using Dynamixels with an 1MB/s speed, I get around 2ms for a packet round trip if I remember correclty.

Can you show the code sample that start the serial port and set the speed and tell me more about your OS?

embeye, 2017-04-23:
My project is under name qembeddedeye from the profile. I can't find a suitable way to share all the code other than this. Can you guide me little about how to send command synchronously to servos. Like instead of saying it two times , in one command operation can be done. This will give me 50 % improvement. I tried controller API but it didn't work

emericg, 2017-05-01:
Hi again,
Before thinking about concurrent motion (which is certainly possible, I'll try to write a small doc to use it with the framework) can you try a few things to see if we have a bug or a setup problem.

- First, do you "see" the servo lagging or is it that these 200ms slow down your program?
- Did you had a look at this page, see if you can find an useful tip?
 https://github.com/emericg/SmartServoFramework/blob/master/doc/SerialCommunication.md
- Can you tell me a bit more about your OS and serial adapter?
- Can you use the GUI in order to see if you see the same delay and validate you are at the rightbaudrate (should be 1 Mbps)
- There is an internal feature to mesure per command latency. If you see the same results, it would definitly point to a serial port issue (either framework bug or misconfiguration). You can enable it in "Dynamixel.cpp" by uncommenting the LATENCY_TIMER define. Then in order to see the result you should change this line (in the same file)
```
    TRACE_1(DXL, "TX > RX loop: %iµs", loopd);
into >
    TRACE_WARNING(DXL, "TX > RX loop: %iµs", loopd);
```

emericg, 2018-06-19:
No activity, closing.

closed 2018-06-19 by emericg