emericg/SmartServoFramework #16: Linux - Two errors compiling GUI and how to fix them

marcodsad, 2017-11-11:
hey there, I'm on Linux Mint 17 with g++ 4.8.
The library compiled fine, but could not compile the GUI. Eventually I managed to successfully compile the GUI by slightly modifying the GUI's main.cpp and makefile.
Here is a summary of the two errors and the respective solutions:

1) ERROR 
usr/include/c++/4.8/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.

SOLUTION: enable c++11 for gcc
- open makefile (in /gui/)
- locate CXXFLAGS
- add -g -std=c++11

2) ERROR 
src/main.cpp:42:22: error: incomplete type ‘QLocale’ used in nested name specifier
     QString locale = QLocale::system().name().section('_', 0, 0);

SOLUTION: add the missing #include 
- open main.cpp (in /gui/src/)
- add #include <QLocale>

emericg, 2017-11-12:
Hi,
The first error should not be happening. There is a CONFIG+=c++11 in the qmake file that should automatically choose the right c++11 enablement flag depending on the compiler/platform used.

I added QLocale as include for your second issue, I guess that was the one missing?

Thanks for your report :-)

marcodsad, 2017-11-13:
Hi, thanks for your reply. Yes, QLocale was missing. 

Strange about CONFIG+=c++11, I cannot find that line in the gui Makefile.

I'm having a different compile error from scons now. I can't figure this one out though.
When compiling the examples I get:

build/DynamixelController.o: In function `DynamixelController::autodetect_internal(int, int)':
DynamixelController.cpp:(.text+0x1859): undefined reference to `ServoX::ServoX(int, int, int)'
collect2: error: ld returned 1 exit status
scons: *** [ex_advance_scanner] Error 1

any hint?

thanks!

emericg, 2017-11-18:
Hi,
The CONFIG+=c++11 tells qmake to choose the right C++11 flags depending on the compiler used. A "-std=c++11" (or close to that) should be added in the Makefile.

I also fixed the scons build system, the new ServoX file was simply missing.

marcodsad, 2017-11-22:
Great, thank you! Will try out the new scons build system asap.

closed 2018-03-08 by emericg