emericg/SmartServoFramework #21: Dynamixel register value errors

orgicus, 2018-07-26:
Hi,

I've recently tried to use the latest version of SmartServoGui with a couple of XH430-W350 servos. Auto mode didn't detect them, but selecting `DXL v2 @` the appropiate baud rate worked.

The issue is some basic functions are ok, for example LED control, but most functions (including setting goal position) do not.

The log displays a list of register value related errors:

```
.[I][  DXL][autodetect_internal] [#1] XH430-W350 servo found!
[I][  DXL][autodetect_internal] [#2] XH430-W350 servo found!
..........................................................................................................................................................................................................................................................
[E][SERVO][updateValue] [#1] updateValue(reg 47 / MODEL_INFORMATION to 0) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 48 / HOMING_OFFSET to 0) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 49 / MOVING_THRESHOLD to 10) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 50 / ACCELERATION_LIMIT to 32767) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 51 / VELOCITY_LIMIT to 280) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 57 / REG_SHUTDOWN to 52) [REGISTER VALUE ERROR] (min: 0 / max: 1)
[E][SERVO][updateValue] [#1] updateValue(reg 28 / GOAL_VELOCITY to 280) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 75 / REG_PROFILE_ACCELERATION to 0) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 76 / REG_PROFILE_VELOCITY to 0) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 26 / GOAL_POSITION to 2345) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 33 / CURRENT_VELOCITY to 0) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 31 / CURRENT_POSITION to 2345) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 80 / REG_VELOCITY_TRAJECTORY to 0) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 81 / REG_POSITION_TRAJECTORY to 2345) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 47 / MODEL_INFORMATION to 0) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 48 / HOMING_OFFSET to 0) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 49 / MOVING_THRESHOLD to 10) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 50 / ACCELERATION_LIMIT to 32767) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 51 / VELOCITY_LIMIT to 280) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 57 / REG_SHUTDOWN to 52) [REGISTER VALUE ERROR] (min: 0 / max: 1)
[E][SERVO][updateValue] [#2] updateValue(reg 28 / GOAL_VELOCITY to 280) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 75 / REG_PROFILE_ACCELERATION to 0) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 76 / REG_PROFILE_VELOCITY to 0) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 26 / GOAL_POSITION to 2345) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 33 / CURRENT_VELOCITY to 0) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 31 / CURRENT_POSITION to 2345) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 80 / REG_VELOCITY_TRAJECTORY to 0) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 81 / REG_POSITION_TRAJECTORY to 2345) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 31 / CURRENT_POSITION to 2345) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#2] updateValue(reg 31 / CURRENT_POSITION to 2345) [REGISTER VALUE ERROR] (min: 0 / max: -2147483648)
[E][SERVO][updateValue] [#1] updateValue(reg 32 / CURRENT_SPEED) [REGISTER NAME ERROR]
[E][  DXL][dxl_print_error] [#1] Protocol Error: Data length shorter than required length for the corresponding address!
[E][SERVO][updateValue] [#1] updateValue(reg 34 / CURRENT_LOAD) [REGISTER NAME ERROR]
[E][  DXL][dxl_print_error] [#1] Protocol Error: Data length shorter than required length for the corresponding address!
[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_serie: '0' / servo_model: '0'
[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_serie: '0' / servo_model: '0'
[E][TABLES][getRegisterTable] Unable to find a suitable 'Control Table' for servo_serie: '0' / servo_model: '0'
```

I had a look at the [Dynamixel Control Tables](https://github.com/emericg/SmartServoFramework/blob/master/SmartServoFramework/ControlTablesDynamixel.h#L321) for this servo model and everything looks according to the Robotis spec.

How can I get the SmartServoGui to play nice with these servos ?

Thank you,
George

emericg, 2018-07-26:
Yeah unfortunately I didn't have any X servo so I had to implement them blindly...

- First, what is the baudrate you set manually ? The auto scan will only try 1Mbps and 57k I believe. I don't think the right control table has been selected by the software.

- Can you build the last commit and tell me what numbers prints in the new trace inside dxl_get_model_infos() ?
`Unknown servo model number '0xsomething'`
By adjusting the correct number in the dxl_get_model_infos() and dxl_get_model_name() you should be able to have better results.

Keep me posted, we'll fix that.

orgicus, 2018-07-29:
Hi @emericg 

Apologies for the delay and thank you so much for the prompt answer.

- I've set the baud rate to 57600 (as that is what the servos I currently have access to have been configured for)
- I've build the latest version, however I can't see that trace: unfortunately the Gui freezes before the call. This are the last messages I see in console before the freeze:

```
>> scanServos(/dev/tty.usbserial-FT2GZFM6)
[I][M-API][updateInternalSettings] - Using Dynamixel communication protocol version 2
[I][SERIAL][SerialPortMacOS] - Device name has been set to: 'tty.usbserial-FT2GZFM6'
[I][SERIAL][SerialPortMacOS] - Device node has been set to: '/dev/tty.usbserial-FT2GZFM6'
[I][SERIAL][SerialPortMacOS] - Device baud rate has been set to: '57600'
```

I did spot the case that should match these servos:
```c++
case 1000:
        servo_serie = SERVO_X;
        servo_model = SERVO_XH430_W350;
```
I'm not sure what `model_number` the frameworks finds.

orgicus, 2019-01-17:
Hi Emeric,

At the time I had to move away from SSF and just manually put the packets together based on the specs. 

One annoying issue was sometimes unplugging/replugging the servo resulted in OSX sometimes not releasing the reference at driver level which meant I always had to reboot before trying again. This made it frustrating to test. I did eventually got to the bottom of the driver lock issue and that was due to the current implementation of the official FTDI drivers which I've completely uninstalled and am using the AppleFTDI driver. I've got confirmation from FTDI staff that currently there's a fault with the drivers and that will be fixed.

Back to SSF, I should be able to get back to getting my hands on Dynamixel X servos and running the debugger on that GUI freeze issue. I know it's a bad practice but I may also pepper the trail from the model number switch statement to the GUI with console prints, at least temporarily.

Apologies I haven't updated you on this issue yet.
I'll do my best to test/update you and fingers crossed even submit a PR at the end of it.

arnaudby, 2019-02-11:
Aside from the suitable control table problem, it seems that there is an error in the max value calculation when a precise range is not specified. This problem goes back here for 4 bytes registers.

emericg, 2019-02-23:
Hi, thanks for the feedback!
I'm not working on the framework at the moment but I'll be sure to look at that when I get back to it.

closed 2021-02-23 by orgicus