emericg/SmartServoFramework #22: SmartServoGui

elthef, 2018-10-01:
I used to be able to run the smart gui program to access the data from the 0402 servo.
However, recently I encountered serial not found problem. Any suggestion ?
Thank you

emericg, 2018-10-02:
Hi,
can you detail what was working before (OS, version of the software) and what changed now that it's not working?

Thanks

elthef, 2018-10-03:
I am working in Ubuntu 16.04 LTS xenial,
Qmake version 3.0
Using QT 5.5.1
The smart GUI interface can  start SmartServoGui but under the San fro new serial poer it reported no serial port.
I have already chmod +x for /dev/ttyUSB0

Edwin

emericg, 2018-10-05:
Hey,
You really should add your user instead of chmoding the serial port device, it will not be enough. The serial port not showing up at all is usually a good sign of just not having permissions to access the device.
# usermod -a -G uucp,dialout $USER

Between the moment where it was working and now, did you changed the version of SmartServoGui, of your serial adapter, or anything?

elthef, 2018-10-05:
I managed to rectify the problem. It seems that I need to uninstall QT5 and reinstall QT4 then recompile the program and it works

closed 2018-10-05 by elthef