emericg/SmartServoFramework #23: GUI Segfault at startup

marcodsad, 2019-01-14:
Hi, I'm on Linux Mint 19.

Compiled both the library and the gui (with qt5). No errors during compilation.
However, as soon as I launch smartservogui (in SmartServoGui/bin) I get a segfault, see below.

servogui 
qt5ct: using qt5ct plugin
Warning: no configuration file, using default values! A new one will be created next time you change a setting.
Parsing settings from fallback string to document succeeded.
Segmentation fault (core dumped)

Am I doing anything wrong?
I used to use a previous version of SSG but that was on Linux 17.

Thanks

emericg, 2019-01-14:
Hi,
in any case, if it segfaults, it's me who did something wrong ^^ I'll have a look at it tomorrow, but in the mean time can you try the appimage version from the release page : https://github.com/emericg/SmartServoFramework/releases

and if you are building the sources, maybe just running the application through GDB and pointing me to what exactly crashed? thanks!

marcodsad, 2019-01-15:
Hi, thanks for the prompt reply.
The appimage works correctly. It starts, connects and finds motors.

Here's the gdb output:

```
(gdb) run
Starting program: /home/...../src/SmartServoFramework-master/SmartServoGui/bin/smartservogui 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffec0ac700 (LWP 9694)]
qt5ct: using qt5ct plugin
[New Thread 0x7fffdf026700 (LWP 9695)]
Warning: no configuration file, using default values! A new one will be created next time you change a setting.
Parsing settings from fallback string to document succeeded.
[New Thread 0x7fffd349c700 (LWP 9696)]

Thread 1 "smartservogui" received signal SIGSEGV, Segmentation fault.
0x0000555555ba9390 in ?? ()
(gdb) 
```

emericg, 2019-01-15:
Humm that's weird.
The gdb trace is empty, are you sure you built the project in debug mode? Otherwise that would point out that the segfault is not in the SmartServoGui code but probably in a loaded library.
There is only 3 commits between the latest AppImage build and master, and two of those are not even code. That would point to something wrong in your build. Is it a clean checkout or something you've had for a while? My advice would be to clean and rebuild everything. I'm downloading a linux mint 19 to try that myself, we never know, but I'm not that hopefull.

If the AppImage is starting at least you can work in the mean time?

marcodsad, 2019-01-16:
yes, I can work for now with the Appimage, thanks for pointing me there.

I always compile from a new checkout, just to be safe. So I did it again, started off with a clean checkout, compilation runs fine (here is a log https://pastebin.com/xCU9fcPr ).

this time gdb throws something:

`(gdb) run
Starting program: /home/sad/src/SmartServoFramework-master/SmartServoGui/bin/smartservogui 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffec0ac700 (LWP 13116)]
qt5ct: using qt5ct plugin
[New Thread 0x7fffdf026700 (LWP 13117)]
Warning: no configuration file, using default values! A new one will be created next time you change a setting.
Parsing settings from fallback string to document succeeded.
[New Thread 0x7fffd349c700 (LWP 13118)]

Thread 1 "smartservogui" received signal SIGSEGV, Segmentation fault.
0x00007ffff799daaa in QTreeView::setRootIndex(QModelIndex const&) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
(gdb)`

emericg, 2019-01-17:
I tried on a new Mint VM and new clone, everything works as intended.
Your backtrace is now showing a segfault inside Qt and not much more, but at least its something ^^ I think you need to type 'bt' in the gdb shell to show more infos about what lead to the crashed function (hopefully this will lead to something in my code, otherwise it would just be some weird Qt crash).

It's a treeview that's crashing, so in the meantime can you try running the app with a serial adapter but without servos, and then without anything plugged at all?

Thanks for your time!

marcodsad, 2019-01-24:
sorry for the delay, I was away for work.
Good to know it's working on your Mint. Perhaps something is wrong with my qt5 install?

Thank you for following up!

Here's the output of bt in gdb *without anything plugged*

`Thread 1 "smartservogui" received signal SIGSEGV, Segmentation fault.
0x00007ffff799daaa in QTreeView::setRootIndex(QModelIndex const&) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
(gdb) bt
#0  0x00007ffff799daaa in QTreeView::setRootIndex(QModelIndex const&) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x00007ffff79312a0 in QAbstractItemView::reset() ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#2  0x00007ffff792e7c9 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#3  0x00007ffff69935b5 in QMetaObject::activate(QObject*, int, int, void**) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00007ffff68ffe4a in QAbstractItemModel::modelReset(QAbstractItemModel::QPrivateSignal) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff6901a0c in QAbstractItemModel::endResetModel() ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff79a5136 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x000055555558f8b2 in MainWindow::scanSerialPorts(bool)
    (this=0x7fffffffdb40, autoScanDevices=true) at src/mainwindow.cpp:270
#8  0x000055555558f7da in MainWindow::autoScanPorts() (this=0x7fffffffdb40)
    at src/mainwindow.cpp:256
#9  0x000055555558da84 in main(int, char**) (argc=1, argv=0x7fffffffdcf8)
    at src/main.cpp:69`

and here the output *with adapter plugged in*
`Thread 1 "smartservogui" received signal SIGSEGV, Segmentation fault.
0x0000555555c74620 in ?? ()
(gdb) bt
#0  0x0000555555c74620 in  ()
#1  0x00007ffff799dab0 in QTreeView::setRootIndex(QModelIndex const&) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#2  0x00007ffff79312a0 in QAbstractItemView::reset() ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#3  0x00007ffff792e7c9 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007ffff69935b5 in QMetaObject::activate(QObject*, int, int, void**) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff68ffe4a in QAbstractItemModel::modelReset(QAbstractItemModel::QPrivateSignal) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff6901a0c in QAbstractItemModel::endResetModel() ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007ffff79a5136 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x000055555558f8b2 in MainWindow::scanSerialPorts(bool)
    (this=0x7fffffffdb40, autoScanDevices=true) at src/mainwindow.cpp:270
#9  0x000055555558f7da in MainWindow::autoScanPorts() (this=0x7fffffffdb40)
    at src/mainwindow.cpp:256
#10 0x000055555558da84 in main(int, char**) (argc=1, argv=0x7fffffffdcf8)
    at src/main.cpp:69`

emericg, 2019-01-24:
Hi, the problem seems to be triggered from:
`void MainWindow::scanSerialPorts(bool autoScanDevices)`
inside mainwindows.cpp, line 259 and beyond

I commited a few more safeguards inside that function but I'm not really hopeful. It crashs inside Qt and from a code section "always used" by everyone launching the application.
You can try to comment some code there to see what line is the root cause, but most likely it's a problem between Qt and your compiler?! Or poor usage of some Qt API that only crash with your setup.

marcodsad, 2019-01-28:
Hi, ok, thanks will try out the new commit.

This is the compiler version I'm using, what's the one you used in your Mint trial?
QMake version 3.1
Using Qt version 5.9.5 in /usr/lib/x86_64-linux-gnu

Since it looks like one of those idiosyncratic bugs which are quite difficult to solve, can I simply work with the App image? What's the difference between image and compiled version?

thanks for the support,

emericg, 2019-01-28:
> gcc version 7.3.0 (Ubuntu 7.3.0-27ubuntu1~18.04) 
> QMake version 3.1
> Using Qt version 5.9.5 in /usr/lib/x86_64-linux-gnu

Same as you I think, it's a new install of Mint 19.1.

For instance the AppImage are built on travis CI for each new commit, and usually kept online for two weeks, unless it's a tagged version (then I upload it on the github release page).
There is usually no "real" differences between them and what you can build on your own computer, only slightly older libraries are used in order to work with most (older) OS.
Latest one : https://transfer.sh/e7QVX/SmartServoGui-git.cc14fa6-linux64.AppImage

emericg, 2019-01-29:
Sorry I couldn't do more.
I'll update you if something comes up!

closed 2019-01-29 by emericg