emericg/SmartServoFramework #25: [gui]Use QSerialPortInfo::availablePorts() for filling out the possible ports list

martonmiklos, 2019-03-21:
Hello @emericg 

First of all many thanks for creating, maintaining and sharing this software, it really really rocks! I appreciate your invested time!

I have came across the GUI first and noticed the following on Linux:
[image]
There are non existing ports (/dev/ttyACM0, cu.usbserial) listed and not all ports listed (I have ttyUSB1 as well).

By reading through the code I realised that these values are coming from the default json settings.

How would it sound for you if we would dynamically fill this widget area by using the Qt's builtin QSerialPortInfo::availablePorts()?

I have seen that you have removed the qtserialport dependency in 3787e190, but I think it would not hurt to move this dependency back since it is part of the Qt since 5.1 and as I have seen the projects depends on Qt5 already. 

If you are open for such an improvement I will came up with a PR.

emericg, 2019-03-23:
Hi!
Thank you for the kind words, I'm happy it works for you!

Regarding the GUI and the port settings saved on disk, yes indeed the initial values are hardcoded, a long time ago I set a selection of what you can find in the wild on various OS in the default settings, then IIRC saved on disk when the application is run for the first time.

Having a dynamic section would be pretty cool indeed. I'm thinking fully dynamic and saved on disk only when the port is "ticked". Right now as I said, the four slots are always saved during the first run.

Good thinking to ask me about it first too ;-)
Qt 4 compatibility was important in 2014 when the software was created, but now we can safely assume everyone is on Qt 5.1 and probably higher. I would be OK to enable it by default for the GUI (retro compatibility and auto detection couldn't hurt, but really not a requirement anymore).
So if you want to implement this feature, yes I'm good with it.

QtSerialPort is mostly working as the other backends, but it has a major flaw that makes it really unusable for the framework: it doesn't work with the "Managed API". Again if I recall correctly, the thread using the serial port must be the one that own the QSerialPort, and must also be a QThread, making the QtSerialPort backend not suitable (at all) as a drop in replacement as I originally hoped. It works great for auto-detection & port locking (pretty convenient and cross platform actually), and also works great for the simple API.

- If you want to re-use a bit of code, I think that the FEATURE_QTSERIAL can be set by default inside the SmartServoGui.pro, and then QtSerialPort would be used by default for SerialPort::scanSerialPorts() and for the locking mechanism, so that would actually benefits other parts of the application too.
Just one very **important thing**, the actual usage of the SerialPortQt backend (for the serial communications) should simply be disabled in both Dynamixel::serialOpen() and HerkuleX::serialOpen(). It's a couple of lines to remove, but other than that, FEATURE_QTSERIAL should be ready.

I'm also pretty sure you can use SerialPort::scanSerialPorts() directly.
I can also see that mainwindow.cpp line 324 can now be factorized like that instead of using the ifdef:
```
// Scan for available/new serial ports
std::vector <std::string> availablePorts = SerialPort::scanSerialPorts();
```

Let me know if that plan sounds good or if simply adding the autodetection with QSerialPortInfo::availablePorts() makes more sense for you.

martonmiklos, 2019-03-23:
Hello!

Many thanks for your detailed answers!

I do not wanted to make the QtSerialport for communication backend I have just been thinking about using it's port enumerator. 

I would have two more questions regarding to this area:
- I think it would be useful to add the address limitations to the settings screens as well (and save it's range to the ports)
- What was the motivation behind the rapidjson based settings loading/saving? I know that QtJSON might not be part of the Qt back in 2014, but I usually use QSettings for such a purpose. begin/endGroups could be chained, so a Json like topology could be used.

emericg, 2019-03-23:
> I do not wanted to make the QtSerialport for communication backend I have just been thinking about using it's port enumerator.

Yes yes I understood, I was talking about using only the enumeration and port locking of the current qt backend, and disabling communication. But code it as you see fit, I'll refactor it after if it make sense.

> I think it would be useful to add the address limitations to the settings screens

Yes definitely

> What was the motivation behind the rapidjson based settings

I'm using QSettings too when I code a new application, but at the time either I didn't know about it or I didn't know how to use it properly. So it's just historical.

martonmiklos, 2019-03-25:
I have started to work on this, made some progress here:
https://github.com/Rescube/SmartServoFramework/commit/5a29e0a2728632ba961374724456d7b4e8d66ce0

I have replaced the rapidjson settings loading with QSettings based one just because I touched the settings loading/save several points. 

I have an open question what I will test tomorrow on Windows (I remember some differences between the portName/systemLocation on Windows/Linux), but more or less my changes are quite ready for a PR.

emericg, 2019-03-28:
Yes it seems good, let me know if you need something.

emericg, 2019-04-27:
Feature has been merged, see pull request https://github.com/emericg/SmartServoFramework/pull/28.
Thanks ;-)

closed 2019-04-27 by emericg