emericg/SmartServoFramework #27: [gui]Minimum Qt version?

martonmiklos, 2019-03-31:
Hi @emericg 

What is the minimum preferred Qt version for the gui?

I saw 5.10 in the Travis config and 5.11 on the appveyor. 

I almost added accidentally some Qt 5.12 only things with the dynamic port config patch: the QtDesigner smuggled some new >5.12 only items to the palettes of the QLCDNumber.

emericg, 2019-03-31:
Well until your pull request it was Qt 4.8 ^^
The CI builders are indeed using recent versions of Qt but that's not the minimum required versions.

If the QtSerialPort has been introduced with Qt 5.1, that should be the new default. Realistically though, I think we'll have trouble testing (or justifying) anything older than 5.6 so would be OK with that version.
I am not aware of new Qt Widget stuff being introduced with recent versions, but if you find some, better avoid them!

martonmiklos, 2019-03-31:
Thanks for the answer I will install an 5.6 for sanity checking. It might be useful to look around

emericg, 2019-04-01:
I would set the CI builds to Qt 5.6 but later versions contains useful stuff like better hdpi support. It's nice to have because I package and use the resulting builds for publication.
I can check if adding one linux build in the build matrix is not too much work. One with an older Qt version (and older compiler why not) and which the output would not be used for packaging.

martonmiklos, 2019-04-02:
Adding a 5.6 to the CI process would be a great thing I think just to make 100% confident that no non-backward compatible code slips in.

martonmiklos, 2019-04-23:
I think I got answer to this question closing now.

closed 2019-04-23 by martonmiklos