emericg/SmartServoFramework #30: Mac not seeing serial port USB2Dynamixel with SmartServoGUI

iseanstevens, 2020-08-04:
The adapter communicates properly with Robotis' DynamixelWizard2.
It shows up as /dev/tty.usbserial-A9501CMS in terminal. 

I've tried launching SSG using sudo, no difference.

Experimented with populating the serial path in SSG preferences.
Keeps showing no serial port available. 

Running macOS Catalina 10.15.6
Apple's built in FTDI drivers.

Thanks let me know if there is anything else I should try. 
Framework seems cool, thanks!

emericg, 2020-08-04:
Hi, I'll have a look and see if something changed recently with new macOS versions!

emericg, 2020-08-09:
Hi, so sorry for the delay.
I have the same problem as you. The API we used to scan for serial ports on macOS doesn't seems to provide results anymore, not sure why, it may need to be rewritten. 

If you are using the framework in an application, you can just provide a port instead of relying on the autodetection. The GUI uses it's own autodetection for persistent port settings, then match it to the results of the framework, that's why adding ports in the settings won't work (if I recall correctly).

In the meantime, you can work around the autodetection directly in the framework code :
in SerialPortMacOS.cpp line 140 or something, you can manually add your own port:
`availableSerialPorts.push_back("/dev/cu.usbmodem14101");
`

emericg, 2020-08-09:
Well actually it might be an easy fix, can you build master or test the latest CI version?

iseanstevens, 2020-08-10:
Thanks! I'm happy to test but don't see where to download a recent CI Mac binary? (which I could be missing something obvious) 
I'll give a quick try to building though usually that requires me beating my head against things for a while...

emericg, 2020-08-10:
No problem, here is the direct link: https://transfer.sh/mA29U/SmartServoGui-git.f1568ff-macOS.zip

iseanstevens, 2020-08-10:
Thanks - So it's still saying "no serial port available". Though in your settings window it does perhaps see there is a port there with the correct "/dev/tty.usbserial-A9501CMS" name filled in, see image.

 I didn't see anything immediately obvious in Console.
Let me know how I can help & thanks for your help!

[image]